stereolabs/zed-open-capture #2: Add support for MacOS

tanzby, 2020-06-06:
I try to build  zed-open-capture on MacOS. All dependencies are satisfied but some Linux headers are not. Such as  'linux/usb/video.h'.   Can you fix that?

```
[  7%] Building CXX object CMakeFiles/zed_open_capture.dir/src/videocapture.cpp.o
/zed-open-capture/src/videocapture.cpp:32:10: fatal error:  'linux/usb/video.h' file not found
#include <linux/usb/video.h>  // for UVC_GET_CUR, UVC_SET_CUR, UVC_GET_LEN
```

Myzhar, 2020-06-06:
Hi @tanzby 
we planned to add "video grabbing support" for MacOS and Windows in the next months, giving priority to the highest number of requests

baidut, 2021-07-02:
@Myzhar  May I ask if it's working on windows now?
Is there an installation tutorial for windows?

Myzhar, 2021-07-04:
@baidut the support for Windows and MacOs has not yet been scheduled.

baidut, 2021-07-08:
Thanks for letting me know!

ycvision, 2022-12-02:
Hi @Myzhar, I'd like to follow up on this thread. Is there any update on the support for MacOS?

Myzhar, 2022-12-02:
I'm sorry, but MacOS support is not planned.

christianbator, 2025-01-28:
I've been working on a macOS version if anyone's interested: [https://github.com/christianbator/zed-open-capture-mac]( https://github.com/christianbator/zed-open-capture-mac).

Right now you can process frames in various resolutions and color spaces, and you can use basic camera controls for one connected ZED camera.

I haven't gotten to sensor input or synchronization yet, but feel free to submit PRs if you'd like!