stereolabs/zed-open-capture #5: [Question] Is the result of algorithm in OpenCV stereo the same as the result in ZED SDK?

violet17, 2020-07-03:
I want to use ZED camera to compute depth map without discrete GPU.
According to the [issue](https://github.com/stereolabs/zed-ros-wrapper/issues/580), I want to get more detail about computing depth map.

Thanks.

Myzhar, 2020-07-03:
Hi @violet17 
the starting point for you is the [rectify example](https://github.com/stereolabs/zed-open-capture/blob/master/examples/zed_oc_rectify_example.cpp).
Once you have the rectified images `right_rect` and `left_rect` you can directly use them with one of the OpenCV stereo matcher to obtain the disparity map:
https://docs.opencv.org/trunk/dd/d86/group__stereo.html

This is an example:
https://resources.oreilly.com/examples/9781786469717/blob/master/OpenCV3ComputerVisionApplicationProgrammingCookbookThirdEdition_Code/Chapter11/stereoMatcher.cpp#L144-173

violet17, 2020-07-06:
@Myzhar   Thanks for the solution. I will have a try.

violet17, 2020-07-07:
> Hi @violet17
> the starting point for you is the [rectify example](https://github.com/stereolabs/zed-open-capture/blob/master/examples/zed_oc_rectify_example.cpp).
> Once you have the rectified images [...]

Hi @Myzhar , Can I directly compute the depth map using this formula： `Z = B*f/disparity`? Do I need to add some scales?
How to get the value of `distance of two cameras B`?

closed 2020-07-16 by violet17