stereolabs/zed-open-capture #9: [Question] AGC ROI

Hakiwen, 2021-01-12:
## Context
Using the ZED SDK, it is possible to set a region of interest (ROI) for AEC/AGC as seen in the [zed-examples](https://github.com/stereolabs/zed-examples/blob/ee3ef19886e0ad2573648e87c0fe49ce03b56009/camera%20control/cpp/src/main.cpp#L211) and detailed in the [API Reference](https://www.stereolabs.com/docs/api/group__Video__group.html#ga7bab4c6ca4fd971055eca1fdd9f4223d).
## Question:
Is it possible to set the ROI for AEC/AGC at the driver level, i.e. using zed-open-capture?

Myzhar, 2021-01-12:
Hi @Hakiwen 
currently, it's not possible to set the ROI for AEG/AGC with the Open Capture driver.
The feature will be added soon.

Myzhar, 2021-03-02:
The ROI feature is ready:
https://github.com/stereolabs/zed-open-capture/releases/tag/v0.3.0

closed 2021-03-02 by Myzhar