stereolabs/zed-open-capture #10: [BUG] Setting frame rate does not work for ZED2 camera

th-lp, 2021-01-21:
Setting the FPS for video capture has not the intended effect. I extended the included example zed_oc_video_example.cpp and set a resolution and fps setting: 

    params.res = sl_oc::video::RESOLUTION::HD720;
    params.fps = sl_oc::video::FPS::FPS_15;
    
    sl_oc::video::VideoCapture cap(params);

While the resolution gets set correctly, the FPS setting does not have the intended effect. When looking at the timing of the incoming images, the frame rate is around 50fps:

...
Time to last frame 22 ms
Time to last frame 23 ms
Time to last frame 19 ms
Time to last frame 23 ms
Time to last frame 22 ms
Time to last frame 20 ms
Time to last frame 22 ms
Time to last frame 25 ms
Time to last frame 17 ms
Time to last frame 22 ms
Time to last frame 23 ms
Time to last frame 20 ms
...

When setting the 
    params.fps = sl_oc::video::FPS::FPS_100;

the frame's arrive faster: 
Time to last frame 17 ms
Time to last frame 16 ms
Time to last frame 16 ms
Time to last frame 16 ms
Time to last frame 16 ms
Time to last frame 17 ms
Time to last frame 15 ms
Time to last frame 17 ms

So setting params.fps seem to have some effect, but not that it sets the FPS setting of the camera to 15 FPS.

**Device information:**
 - OS: Ubuntu
 - OS version: 20.04
 - CPU: AMD Ryzen 7 1700 Eight-Core Processor
 - GPU Nvidia:  GeForce GTX 1070
 - ZED SDK Version: 3.3

**To Reproduce**
Steps to reproduce the behavior:
Use the example I described above and print out the frame timings

**Expected behavior**
Frames should arrive with 15 FPS.

th-lp, 2021-01-21:
My ZED Diagnostics output:
{
    "Camera Test": {
        "ZEDCount": 1,
        "deviceInitResult": 0,
        "firmware": 1523,
        "resolutions": {
            "HD1080@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1080,
                "height (output)": 1080,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD1080@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 1080,
                "height (output)": 1080,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1920,
                "width (output)": 1920
            },
            "HD2K@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 1242,
                "height (output)": 1242,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 2208,
                "width (output)": 2208
            },
            "HD720@15": {
                "fps (input)": 15,
                "fps (output)": 15,
                "height (input)": 720,
                "height (output)": 720,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@30": {
                "fps (input)": 30,
                "fps (output)": 30,
                "height (input)": 720,
                "height (output)": 720,
                "initCode": true,
                "initialization": "Ok",
                "width (input)": 1280,
                "width (output)": 1280
            },
            "HD720@60": {
                "dirty lenses": 3,
                "dropCount": 0,
                "fps (input)": 60,
                "fps (output)": 60,
                "height (input)": 720,
                "height (output)": 720,
                "imageCount": 501,
                "initCode": true,
                "initialization": "Ok",
                "tearingCount": 0,
                "width (input)": 1280,
                "width (output)": 1280
[... 141 lines cut ...]
                "idVendor": "0x413c"
            },
            {
                "USB_path": "/1",
                "idProduct": "0x0005",
                "idVendor": "0x1bcf"
            }
        ],
        "USBMode": 3,
        "ZED Camera Module Detected": 63360,
        "ZED MCU Module Detected": 63361,
        "ZEDDetected": true,
        "valid": [
            "<b>USB Bandwidth: </b> OK"
        ]
    },
    "ZED SDK": {
        "CUDA Toolkit version": "V11.0.194",
        "DiagnosticZEDSDKVersion": "3.3.0",
        "InstalledZEDSDKVersion": "Unknown",
        "SMI": "(SMI) :     Performance State                     : P2",
        "initResult": 0,
        "loaded": true
    }
}

th-lp, 2021-01-21:
Let me add that setting the FPS on the same system and camera works in the ZED Explorer and when using OpenCV as interface. Therfore, I suspect its an implementation problem in ZED-open-caputre instead of a hardware/driver problem.

th-lp, 2021-02-01:
@Myzhar any news on this? Thanks!

Myzhar, 2021-02-01:
It's scheduled for this week. Stay tuned

th-lp, 2021-02-02:
Great. Thanks a lot. Please let me know if you need any more input from my side.

Myzhar, 2021-02-04:
@th-lp I'm testing the FPS issue, but I'm not facing the same problem.
Can you please post the code that you use to calculate the FPS?

Myzhar, 2021-02-04:
@th-lp OK, I found a little bug in the declaration of the `getLastFrame` function:
the default timeout value (10 msec) was too strict and this generated a lot of "not valid" frames.

The frequency of the valid frames was correct, but without comparing the timestamps of the frames to check if the received frame was new, you could not be aware of it.

Now with the correct default timeout value, the function returns only valid frames (unless something wrong happens with the connection of the camera).

You can checkout the branch [fix_fps_issue_#10](https://github.com/stereolabs/zed-open-capture/tree/fix_fps_issue_%2310) and recompile to verify if this works correctly for you.
I slightly modified the `zed_oc_video_example.cpp` example to calculate the FPS in two different ways, you can enable simply removing the comment here:
https://github.com/stereolabs/zed-open-capture/blob/fix_fps_issue_%2310/examples/zed_oc_video_example.cpp#L32

th-lp, 2021-02-08:
@Myzhar thanks for the fix. I tested it and it solves the problem for me.

Cheers,
Thomas

Myzhar, 2021-02-08:
Good. Merged into master. I close the issue

closed 2021-02-08 by Myzhar