stereolabs/zed-open-capture #18: [Question] How to get the synchronized IMU and image data?

chengfzy, 2021-08-19:
I use this SDK to record IMU and image data, and found the timestamp betwwen the IMU and image are not synchronized correctly? Is there example to explain how to obtain the synchronized record?

Myzhar, 2021-08-19:
IMU data run at 400 Hz that is faster than the maximum rate of the image data.
We provide an example that explains how to get the IMU data sample nearest to each image frame:
https://github.com/stereolabs/zed-open-capture/blob/master/examples/zed_oc_sync_example.cpp

chengfzy, 2021-08-20:
Does the syncrhonization service will modify the system(PC) time clock?

Myzhar, 2021-08-20:
No, it doesn't. We do not modify the system clock

chengfzy, 2021-08-25:
How to syncrhonize the IMU(or Camera) timestamp to system(PC) timestamp, I add system timestamp in IMU data, and found obvious gap between them
[image]

Myzhar, 2021-08-25:
Here the problem is when you get the system timestamp with respect to IMU timestamp.
The IMU timestamp is set as soon as the IMU data "reaches the grabber", and it's processed in order to fix it with the timestamp reported by the IMU sensor itself. 
I suppose that you are getting the system timestamp just before or after calling the `getLastIMUData` function... so a little latency is expected and it is expected to be variable data.

chengfzy, 2021-08-25:
I don't getting the system timestamp after calling the `getLastIMUData` function. I modify some code in SDK, after the IMU Raw data is obtained in [https://github.com/stereolabs/zed-open-capture/blob/master/src/sensorcapture.cpp#L371](https://github.com/stereolabs/zed-open-capture/blob/master/src/sensorcapture.cpp#L371), I use `getWallTimestamp` to  record the system timestamp, then add to imu data. This plot is just the difference between sensor timestamp and the system timestamp.

Myzhar, 2021-08-25:
OK, indeed the IMU timestamp and the system timestamp drift in time, that's why a scaling factor must be applied (L393-L394)

chengfzy, 2021-08-31:
I found the norm of acceleration in 3 axies if not equal, or approximat to the graivity(9.81), the calculated value is large than 10, Is the acc scale setting right? The conclusion is the same if using ZED Sensor Viewer. BTW, what't the difference between this library and official SDK with CUDA, does the official SDK also using this library for obtaining sensor data?

Myzhar, 2021-10-20:
The data provided by the camera sensors are the same using the SDK and the Open Capture driver.
The only difference is that the SDK applies calibration filters (if you retrieve calibrated data), while the Open Capture driver outputs only raw data.

closed 2022-04-27 by github-actions[bot]