stereolabs/zed-open-capture #21: Python version

LulaSan, 2021-09-14:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hi, can you add also the python version of this repository?

Myzhar, 2021-09-14:
Hi @LulaSan 
the Python version is not currently provided. There are no plans for it.

AntonioViscomi, 2024-05-09:
@Myzhar Any news? I need to use a ZED2 on a robot that does not have an Nvida gpu and having never used C++ I am somewhat at a loss. It would also be nice to have the ROS2 wrapper for open-capture as well!

Myzhar, 2024-05-09:
Hi @AntonioViscomi 
Apologies, both features you are waiting for are not currently planned in our roadmap.