stereolabs/zed-open-capture #26: use with zed SDK

oskarmue, 2022-07-08:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Is it possible to retrieve the Camera data with this package on one pc (without GPU) and then process it on another pc, on which the the ZED SDK is installed, in real time?

### Use-Case

we have a smaller pc which is connected to all our sensors and then there is a second pc which receives the data from the low level pc. So in order to use the ZED Camera it would be nice to receive the zed camera data on one pc and then process it with the given ZED SDK on the second pc.

closed 2022-07-08 by Myzhar

abroun, 2024-08-29:
Hi @Myzhar, I'd like to check, is this situation still the same as of now, 2024? I ask because we have a similar use case, Raspberry Pi powered device, streaming camera and sensor data to a CUDA capable laptop, and we're currently evaluating whether a Zed 2 would be a good fit for our application.

I can see from the samples that it's possible to pull the raw data from a Zed 2, and then later rectify the images and calculate the depth maps, but can you please comment on how the quality of these depth maps will compare to those produced by the SDK? Is there anything major that we'll be missing out on by not being able to process the camera data with the SDK?

Thanks in advance for your time.

Regards

Alan

abroun, 2024-08-29:
Thanks for the info @Myzhar. Much appreciated.