stereolabs/zed-open-capture #29: Two ZED 2i camera - Green hue in one half of the images

TR-Monash, 2022-11-04:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am trying to capture 720p stereo images (2560x720) from two ZED 2i cameras using a Raspberry pi 4. While images from one camera seems to be fine, one half of the images captured from the other camera displays a prominent green hue. Please see below for an example. This green hue is present initially in the first half as well. However, it gradually subsides. 
Frame 1
[image]

Frame 3
[image]

Frame 5
[image]

Frame 8
[image]

Frame 10
[image]

This is not an issue with the resolution as the same phenomenon can be observed in the VGA resolution as well. This is observed in all the the ZED 2i cameras I have access to. This is also the case when I ran this on my laptop with Ubuntu 16.04. 

### Steps to Reproduce

1. Connect two ZED 2i cameras to the host 
2. Compile and run the following code to record 25 images from each camera.
3. Observe the green hue problem explained above in the images saved to '/home/USERNAME/Data_Collection/Images/Second" directory.

```
// ----> Includes
#include "videocapture.hpp"
#include "sensorcapture.hpp"
 
#include <unistd.h> // for usleep
#include <iostream>
#include <iomanip>
#include <iomanip>
#include <thread>
#include <fstream>
#include <cstdlib>
#include <mutex>
#include <sys/time.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <errno.h>

#include <opencv2/opencv.hpp>
// Functions

bool displayData = false;

std::string base_path;
 
// The main function
int main(int argc, char *argv[])
{
    char TimeString[128];

    if (argc == 2)
    {
	base_path = argv[1];
	std::cout << "Base path : " << base_path << std::endl;
    }
    else
    {
	std::string userName = std::getenv("USER");
	//static int seconds_last = 99;
[... 251 lines cut ...]
        }
    }

    return EXIT_SUCCESS;
}
```

### Expected Result

Expect both image sets not to have any green hue.

### Actual Result

As mentioned before, images captured from one camera has a prominent green hue in one half of it, while the other camera is functioning just fine.

### ZED Camera model

ZED2i

### Environment

```shell
OS: Raspian, Ubuntu 16.04
RAM: 4GB, 16GB
```

Myzhar, 2022-11-04:
Hi @TR-Monash 
this seems to be a white-balance problem.
Have you tried to record more than 25 frames to see if it disappears?
25 frames at 15 FPS is a bit more than 2 seconds, in some conditions, it cannot be enough for the white-balance ISP algorithm to correctly tune the colors.

Have you tried to open the camera with a webcam viewer (cheese, guvcview, ...) to verify if the problem is present with it?

TR-Monash, 2022-11-04:
Hi @Myzhar,

Thanks for the response. I thought it's white balance issue at first as well. I captured over thousand images and still the issue is there. I specifically set the white balance to Auto in my code.
```    
    videoCapCam1.setAutoWhiteBalance(true);
    videoCapCam1.setAECAGC(true);
    
    videoCapCam2.setAutoWhiteBalance(true);
    videoCapCam2.setAECAGC(true);
```

I just tried with guvcview and I don't see the issue with it. Could this be due to memory leak in one of the YUV channels?

Myzhar, 2022-11-04:
If the problem is not present with `guvcview` then there is something wrong with your code.
Have you tested `guvcview` in a multi-camera configuration?

TR-Monash, 2022-11-04:
I tested `guvcview` in a multi-camera configuration. I believe, two video captures are operating in two different threads in `guvcview`.

TR-Monash, 2022-11-04:
I tried to create a minimal working example and now it seems to be working just fine! I'll close the issue for now and try to debug and see what caused the issue. Thanks for your input.

TR-Monash, 2022-11-04:
Hi @Myzhar,
Unfortunately, I was wrong. What has happened is that `guvciew` has "reset" the cameras somehow and when I use them back with my code, everything is working just fine. However, if I disconnect and reconnect the cameras, I can observe the above described phenomenon. Again, when I disconnect and reconnect the cameras and then capture video using `guvciew`, the issues are gone. I'm clueless as to what is happening here. 
For clarity, here is a minimal working code to reproduce the issue.
```
// ----> Includes
#include "videocapture.hpp"
#include "sensorcapture.hpp"
 
#include <unistd.h> // for usleep
#include <iostream>
#include <iomanip>
#include <iomanip>
#include <thread>
#include <fstream>
#include <cstdlib>
#include <mutex>
#include <sys/time.h>
#include <sys/types.h>
#include <sys/stat.h>
#include <errno.h>

#include <opencv2/opencv.hpp>

bool displayData = false;

std::string base_path;

// The main function
int main(int argc, char *argv[])
{
	
	std::string userName = std::getenv("USER");
	base_path = "/home/"+ userName + "/Data_Collection/" ;//+ std::string(TimeString);

    
    std::cout << "Base path cpp : " << base_path << std::endl;

    
    // Silence unused warning
 
    // Set the verbose level
    sl_oc::VERBOSITY verbose = sl_oc::VERBOSITY::INFO;
    
    // Set the video parameters
    sl_oc::video::VideoParams params;
    params.res = sl_oc::video::RESOLUTION::HD720;
    params.fps = sl_oc::video::FPS::FPS_15;
    params.verbose = verbose;
 
    // Create a Video Capture objects
    sl_oc::video::VideoCapture videoCapCam1(params);
    if( !videoCapCam1.initializeVideo(0) )
    {
        std::cerr << "Cannot open camera video capture for camera 1" << std::endl;

        return EXIT_FAILURE;
    }
    
    sl_oc::video::VideoCapture videoCapCam2(params);
    if( !videoCapCam2.initializeVideo(2))
    {
        std::cerr << "Cannot open camera video capture for camera 2" << std::endl;

        return EXIT_FAILURE;
    }
    
    // Set video parameters
    videoCapCam1.setAutoWhiteBalance(true);
    videoCapCam1.setAECAGC(true);
    
    videoCapCam2.setAutoWhiteBalance(true);
    videoCapCam2.setAECAGC(true);
	
    int w,h;
    videoCapCam1.getFrameSize(w,h);
    
    cv::Mat frameBGR(h, w, CV_8UC3, cv::Scalar(0,0,0));
    cv::Mat frameBGR2(h, w, CV_8UC3, cv::Scalar(0,0,0));

    uint64_t last_timestamp = 0;

    // Infinite grabbing loop
    std::string rear_image_path = base_path +"/Images/Rear/" + "Rear_";
    std::string front_image_path = base_path +"/Images/Front/" + "Front_";
    std::cout << "Start capturing data" << std::endl;
    while (1)
    {
        // Get Video frame
        // Get last available frame
        const sl_oc::video::Frame frame1 = videoCapCam1.getLastFrame(1);
	const sl_oc::video::Frame frame2 = videoCapCam2.getLastFrame(1);
	
        if(frame1.data!=nullptr && frame1.timestamp!=last_timestamp)
        {
            // Conversion from YUV 4:2:2 to BGR for visualization
            cv::Mat frameYUV( frame1.height, frame1.width, CV_8UC2, frame1.data);
            cv::cvtColor(frameYUV,frameBGR, cv::COLOR_YUV2BGR_YUYV);
	    
	    std::string image_timestamp = std::to_string(frame1.timestamp);
            cv::imwrite(rear_image_path + image_timestamp + ".jpg", frameBGR);
        }
	if(frame2.data!=nullptr)
        {
            // Conversion from YUV 4:2:2 to BGR for visualization
            cv::Mat frameYUV2( frame2.height, frame2.width, CV_8UC2, frame2.data);
            cv::cvtColor(frameYUV2,frameBGR2, cv::COLOR_YUV2BGR_YUYV);
			
            std::string image_timestamp = std::to_string(frame1.timestamp);
            cv::imwrite(front_image_path + image_timestamp + ".jpg", frameBGR2);
        }

        // Keyboard handling
        int key = cv::waitKey(1);

        if( key != -1 )
        {
            // Quit
            if(key=='q' || key=='Q'|| key==27)
            {
                break;
            }
        }
    }

    return EXIT_SUCCESS;
}
```

Myzhar, 2022-11-04:
Hi @TR-Monash 
I was able to replicate the issue on my host PC, but I have not found the cause of the problem.
I'm going to investigate deeper into it and provide a solution as fast as possible.

Myzhar, 2022-11-04:
@TR-Monash I just pushed a new version in the master branch with a multicamera example that seems to be not affected by this issue.
Can you test it?
You can use the "a" or "A" key to enable/disable auto gain/exposure and auto white balance while the example runs.

TR-Monash, 2022-11-04:
@Myzhar 
Thank you very much for the sample code and I'll try it today and report back.
In the meantime, I was able to get my minimal code working by commenting the automatic gain/exposure control, interestingly.
```
videoCapCam1.setAECAGC(true);
videoCapCam2.setAECAGC(true);
```

TR-Monash, 2022-11-05:
@Myzhar 
I can confirm that the example code you have pushed is working as expected.
I can't see any major difference between that and the the minimal code sample I have shared though. I'll deep dive into that as time permits. I'll adopt your code for the time being.
Thank you.

Myzhar, 2022-11-06:
@TR-Monash I guess it's some USB communication conflict that makes the camera FW fail.
We will investigate it further trying to understand the conditions that make it happen

TR-Monash, 2022-11-07:
@Myzhar 
Thank you very much. I'll close this issue for now as the example code works as intended. 
Please post an update once you figure out what's causing this.

closed 2022-11-07 by TR-Monash