stereolabs/zed-open-capture #30: segfault  when used as a library

Hi-Zed, 2022-12-01:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

Hello,
I am trying to implement a ROS wrapper for this library to use my ZED2 on a machine without GPU.
Everything was working fine until I tried to publish a CameraInfo message.
As soon as I add the following instruction `info_left.header.stamp = ros::Time::now();` (or any instruction that changes the time value of the header) I get a segmentation fault.
If I run the code in debug, the segmentation fault is raised somewhere in `sl_oc::sensors::SensorCapture::updateTimestampOffset`, which is even more surprising, since that method should not be called, since I did not set `SENSORS_MOD_AVAILABLE`.

In an effort to try to understand which instruction was causing the problem, I imported the entire source code of the library into my ROS package. When compiled this way, I get no segmentation fault and the node works with no problems.

### Steps to Reproduce

1. Compile and install zed-open-capture
2. Create a simple ROS package with the following source
```
#include <ros/ros.h>
#include <image_transport/image_transport.h>
#include "videocapture.hpp"

int main(int argc, char** argv)
{
    sl_oc::video::VideoParams params;
    params.res = sl_oc::video::RESOLUTION::HD2K;
    params.fps = sl_oc::video::FPS::FPS_15;

    sl_oc::video::VideoCapture cap(params);
    cap.initializeVideo(-1);

    ros::init(argc, argv, "image_publisher");
    ros::NodeHandle nh;

    sensor_msgs::CameraInfo info_left;
    info_left.header.stamp = ros::Time::now();

    ros::spin();
}
```
This is the `CMakeLists.txt` I am using
```
cmake_minimum_required(VERSION 3.0.2)
project(zed_open_capture_ros)

find_package(catkin REQUIRED COMPONENTS
        image_transport
        sensor_msgs
        std_msgs
        )

catkin_package(
)
include_directories(
         include
        ${catkin_INCLUDE_DIRS}
        "/usr/local/include/zed-open-capture"
)

link_directories(zed_open_capture "/usr/local/lib")
add_executable(segfault src/segfault.cpp)
target_compile_definitions(segfault PUBLIC VIDEO_MOD_AVAILABLE)
target_link_libraries(segfault ${catkin_LIBRARIES} zed_open_capture)
```
4. Compile with `catkin_make`
5. Run using `rosrun zed_open_capture_ros segfault`
6. Segmentation fault

### Expected Result

All the elements are correctly initialized, and there should be no segmentation fault.
In fact, the node does not trigger a segmentation fault when the library is imported into the package.

### Actual Result

Segmentation fault.

### ZED Camera model

ZED2

### Environment

```shell
My host OS is Ubuntu 22.04, I am developing and running everything in a container with Ubuntu 20.04 and ROS Noetic.
For testing, I am using a ZED2
```

Myzhar, 2022-12-01:
Hi @Hi-Zed 
we noted the issue. 
I will test it in the next few days to fix the problem.

Myzhar, 2023-01-01:
Planned (to not close).

ab3nd, 2023-01-11:
I'm doing literally the exact same thing (ROS1 wrapper for a non-CUDA system) and getting the exact same fault in the same place when I try to update the CameraInfo before posting the messages. 

My code initially loads the camera calibration information, and saves it in two CameraInfo messages, l_info and r_info. Once it's in the capture loop, what it was trying to do was update the timestamps on l_info and r_info and then ship them. Why this crashes the sl_oc::sensors::SensorCapture::updateTimestampOffset function in a different thread is still obscure to me. 

However, if you create a sensor_msgs::CameraInfoPtr to the camera info, update the stamps and frame_ids through that, and then publish the CameraInfoPtr, things appear to operate smoothly. 

```c++
    while(ros::ok)
    {
        const sl_oc::video::Frame frame = cap.getLastFrame();
        if (frame.data != nullptr)
        {
            // Image is YUV 4:2:2, convert to rgb
            cv::Mat frameYUV = cv::Mat(frame.height, frame.width, CV_8UC2, frame.data);
            cv::Mat frameBGR;
            cv::cvtColor(frameYUV, frameBGR, cv::COLOR_YUV2BGR_YUYV);
            // Split into two images
            left_raw = frameBGR(cv::Rect(0, 0, frameBGR.cols / 2, frameBGR.rows));
            right_raw = frameBGR(cv::Rect(frameBGR.cols / 2, 0, frameBGR.cols / 2, frameBGR.rows));
            
            // Create ROS message and publish it
            sensor_msgs::ImagePtr l_msg = cv_bridge::CvImage(std_msgs::Header(), "bgr8", left_raw).toImageMsg();
            sensor_msgs::ImagePtr r_msg = cv_bridge::CvImage(std_msgs::Header(), "bgr8", right_raw).toImageMsg();
            
            ros::Time now = ros::Time::now();
            l_msg->header.stamp = now;
            r_msg->header.stamp = now;
            r_msg->header.frame_id = "zed_right";
            l_msg->header.frame_id = "zed_left";
            
            sensor_msgs::CameraInfoPtr l_info_ptr = boost::make_shared<sensor_msgs::CameraInfo>(l_info);
            sensor_msgs::CameraInfoPtr r_info_ptr = boost::make_shared<sensor_msgs::CameraInfo>(r_info);

            l_info_ptr->header.stamp = now;
            r_info_ptr->header.stamp = now;
            l_info_ptr->header.frame_id = "zed_left";
            r_info_ptr->header.frame_id = "zed_right";

            l_image_pub.publish(l_msg);
            l_info_pub.publish(l_info_ptr);
            r_image_pub.publish(r_msg);
            r_info_pub.publish(r_info);
        }
        ros::spinOnce();
    }
```

Myzhar, 2023-01-11:
I suppose it's a udev rule issue.
The sensor module cannot access the sensors data and for some reason (to be verified) it crashes.
In the `udev` folder you can find the udev rules file and a script to install it

closed 2023-02-16 by github-actions[bot]