stereolabs/zed-open-capture #31: missing cxxapi.h

bobdavies2000, 2023-01-03:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I built hidapi, pointed the cmake for zed-open-capture to it, then pointed zed-open-capture cmake to my opencv 4.7, then configured and generated the zed-open-capture project successfully.  I went to build the resulting project and it fails in several modules with a missing cxxapi.h file.  I could probably repair this but I decided you should know that this happens with your current configuration.  I will be glad to try anything you suggest.

Bob Davies

### Steps to Reproduce

1. Described above.
2. 
3.
...

### Expected Result

I was hoping it would build...

### Actual Result

zed-open-capture will not build as current configured.

### ZED Camera model

ZED

### Environment

```shell
Latest version of zed-open-capture, Windows 10, NVIDIA GeForce RTX 3050 (driver 30.0.15.1179, OpenCV 4.7, ZED 2 camera
```

### Anything else?

No rush to fix this.  I currently work with an older Stereolabs SDK from 2020 but I wanted to review this capture driver and switch to it.

Myzhar, 2023-01-03:
@bobdavies2000 this repository is designed for Linux OS. It cannot be built under Windows.

bobdavies2000, 2023-01-03:
Thanks for the prompt response.

closed 2023-01-10 by github-actions[bot]