stereolabs/zed-open-capture #33: Support for ZED X and ZED X Mini?

peds, 2023-04-06:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Are there plans to support the new ZED X and ZED X Mini?

### Use-Case

Control and access sensor data for the new ZED X and ZED X Mini without having to use the ZED SDK (which requires CUDA).

### Anything else?

Thank you for this awesome library!

Myzhar, 2023-04-06:
This is not currently possible because the Open Capture driver is only for USB3 devices.

ZED X and ZED X Mini are GMSL2 devices and they can be used only with NVIDIA Jetson embedded boards that are CUDA-enabled, so a driver for not CUDA-enabled devices is not useful.

Myzhar, 2023-04-06:
Before closing the issue definitely, @peds can you provide me with more information regarding the host device that you want to use the ZED X with?

peds, 2024-04-03:
> Before closing the issue definitively, @peds can you provide me with more information regarding the host device that you want to use the ZED X with?

Sorry I just noticed this, but the feature would still be useful to me. The host device will be Jetson AGX Xavier Series.