stereolabs/zed-open-capture #35: Inappropriate ioctl for device

apnogueira, 2023-06-27:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I'm just trying to run the examples after installing this library on a Raspberry Pi 4.
I'm encountering errors specifically when running examples using the cameras. The IMU example (zed_open_capture_sensors_example) runs fine.
It appears to be an error while opening the cameras due to the resolution, but I can't figure out how to solve it.
My problem seemed different enough from #19 as I wasn't experiencing any freezes and the camera seemed to be recognized using `lsusb`.

I should note I am not using the original USB type-C cable, as I currently don't have access to it. This is the output of `lsusb`:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 008: ID 2b03:f881 STEREOLABS ZED-2i HID INTERFACE
Bus 001 Device 007: ID 2b03:f880 STEREOLABS ZED 2i
Bus 001 Device 006: ID 0424:2512 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

### Steps to Reproduce

I'm following the install and build instructions in the README.md file. 
After that, I try to run the lines `zed_open_capture_sync_example` and `sudo zed_open_capture_video_example` unsuccessfully. 

### Expected Result

I'm expecting the normal behavior of using the library's own examples.behavior

### Actual Result

When running `sudo zed_open_capture_video_example` (I'm using `sudo` because if I don't, it replies with `Permission denied`)
```
[sl_oc::video::VideoCapture] ERROR: Error setting the camera resolution
xioctl: Inappropriate ioctl for device
xioctl: Invalid argument
[sl_oc::video::VideoCapture] ERROR: Cannot set pixel format of '/dev/video1': [22] Invalid argument
Cannot open camera video capture
See verbosity level for more details.
xioctl: Invalid argument
```

I edited the file `zed_oc_sync_example.cpp` and set `VERBOSITY` to `INFO` and got this result when running `zed_open_capture_sync_example` (no `sudo` required this time):
```
[sl_oc::video::VideoCapture] INFO: ZED Open Capture - Camera module - Version: 0.6.0
[sl_oc::video::VideoCapture] INFO: Camera resolution: 2560x720@30Hz
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video0'
[sl_oc::video::VideoCapture] ERROR: Cannot open '/dev/video0': [13] Permission denied
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video1'
[sl_oc::video::VideoCapture] ERROR: Cannot open '/dev/video1': [13] Permission denied
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video2'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video2/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video2' is not a Stereolabs camera
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video3'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video3/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video3' is not a Stereolabs camera
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video4'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video4/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video4' is not a Stereolabs camera
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video5'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video5/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video5' is not a Stereolabs camera
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video6'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video6/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video6' is not a Stereolabs camera
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video7'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video7/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video7' is not a Stereolabs camera
[... 158 lines cut ...]
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video60' is not a Stereolabs camera
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video61'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video61/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video61' is not a Stereolabs camera
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video62'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video62/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video62' is not a Stereolabs camera
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video63'
[sl_oc::video::VideoCapture] WARNING:  Not a modalias : /sys/class/video4linux/video63/device/modalias
[sl_oc::video::VideoCapture] WARNING: The device '/dev/video63' is not a Stereolabs camera
Cannot open camera video capture
Try to enable verbose to get more info
```

### ZED Camera model

ZED2i

### Environment

```shell
Raspberry Pi 4 Model B Rev 1.2
OS: Raspbian 11 (bullseye)
ZED Open Capture - Camera module - Version: 0.6.0
```

Myzhar, 2023-06-27:
Hi @apnogueira 
Did you install the udev rules?

apnogueira, 2023-06-27:
Hi @Myzhar 
Yes, I did. When I tried to run the command `bash install_udev_rule.sh` I got a bunch of `Permission denied`.
I then changed the command to `sudo bash install_udev_rule.sh` and it didn't show the `Permission denied` (As a matter of fact it had no response to the command). I assumed it installed the udev rules

Myzhar, 2023-06-28:
I guess your first issue is caused by a resolution that is too high.

Did you connect the camera to a USB3 or USB2 port?

Try to change the resolution to VGA to see if you can open the camera (please use `sudo`).

Please also copy and paste the output of the commands `lsusb` and `lsusb -d 2b03: -v -t`

apnogueira, 2023-06-28:
Currently the camara is connected to USB3, but I previously tried all ports (both RPi4 USB3 ports and both USB2 ports). 
I changed both examples (sync and video_example) to a VGA resolution. The error changed.
When running `sudo zed_open_capture_video_example`
```
Connected to camera sn: 39263373[/dev/video0]
No protocol specified
Unable to init server: Could not connect: Connection refused
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.1) ../modules/highgui/src/window_gtk.cpp:624: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'

Aborted
```

When running `sudo zed_open_capture_sync_example`:
```
[sl_oc::video::VideoCapture] INFO: ZED Open Capture - Camera module - Version: 0.6.0
[sl_oc::video::VideoCapture] INFO: Camera resolution: 1344x376@30Hz
[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video0'
[sl_oc::video::VideoCapture] INFO: Opened camera with SN: 39263373
[sl_oc::video::VideoCapture] INFO: Device '/dev/video0' opened
Video Capture connected to camera sn: 39263373
[sl_oc::sensors::SensorCapture] INFO: ZED Open Capture - Sensors module - Version: 0.6.0
[sl_oc::sensors::SensorCapture] INFO: Device Found: 
  VID: 2b03 PID: f881
  Path: 0001:0005:00
  Serial_number:   39263373
  Manufacturer:   STEREOLABS
  Product:   ZED-2i HID INTERFACE
  Release number:   v3.9
***

[sl_oc::sensors::SensorCapture] INFO: Connected to device with sn 39263373
Sensors Capture connected to camera sn: 39263373
No protocol specified
Unable to init server: Could not connect: Connection refused
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.1) ../modules/highgui/src/window_gtk.cpp:624: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem'

Aborted
```

Output of `lsusb`
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 005: ID 2b03:f881 STEREOLABS ZED-2i HID INTERFACE
Bus 001 Device 004: ID 2b03:f880 STEREOLABS ZED 2i
Bus 001 Device 003: ID 0424:2512 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

Output of `lsusb -d 2b03: -v -t`
```
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    ID 1d6b:0003 Linux Foundation 3.0 root hub
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    ID 1d6b:0002 Linux Foundation 2.0 root hub
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        ID 2109:3431 VIA Labs, Inc. Hub
        |__ Port 2: Dev 3, If 0, Class=Hub, Driver=hub/2p, 480M
            ID 0424:2512 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
            |__ Port 1: Dev 4, If 0, Class=Video, Driver=uvcvideo, 480M
                ID 2b03:f880 STEREOLABS 
            |__ Port 1: Dev 4, If 1, Class=Video, Driver=uvcvideo, 480M
                ID 2b03:f880 STEREOLABS 
            |__ Port 2: Dev 5, If 0, Class=Human Interface Device, Driver=, 12M
                ID 2b03:f881 STEREOLABS 

```

Myzhar, 2023-06-28:
The camera is now correctly working, but you have problems with OpenCV displaying the images.

I suggest you search the internet for this error:
`what():  OpenCV(4.5.1) ../modules/highgui/src/window_gtk.cpp:624: error: (-2:Unspecified error) Can't initialize GTK backend in function 'cvInitSystem`

apnogueira, 2023-07-02:
I figured the problem was related with using `sudo`. However, I had no permission to open the camera without `sudo`. I had to include my user in the video group in order to run the example without `sudo`. The command I used was:
```
sudo usermod -aG video USERNAME
```

So right now I am able to run all examples on the RPi4, but only with VGA resolution. Any suggestions on how to use `HD720` resolution?

Myzhar, 2023-07-03:
Hi @apnogueira 
You are using a ZED 2i. Please check that the screws of the connector are strongly strengthened and that you are connecting the camera so a USB3 port.
The camera is detected in USB2 mode for compatibility when there is a connection issue.
If you want to be sure that the camera is working correctly you can try to connect it to a PC, open the video stream with a normal webcam application and try to change the resolution.
If the PC is CUDA-enabled you can install the ZED SDK and run the ZED Diagnostic tool to check that the camera is OK.

apnogueira, 2023-07-06:
Hi @Myzhar 
I finally managed to get my hands on the original cable. With it, I no longer have the resolution issue. I was able to run the examples for HD720 resolution. Thank you! 

Closing...

closed 2023-07-06 by apnogueira