stereolabs/zed-open-capture #36: Why only support one resolution option of VGA?

zhukao, 2023-06-30:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I can only get imgs success with the resolution option of RESOLUTION::VGA!

### Steps to Reproduce

1. set params in `examples/zed_oc_video_example.cpp`
`
    sl_oc::video::VideoParams params;

    params.res = sl_oc::video::RESOLUTION::HD720;

    params.fps = sl_oc::video::FPS::FPS_30;

    params.verbose = sl_oc::VERBOSITY::INFO;
`

2. build and run `sudo ./zed_open_capture_video_example`

...

### Expected Result

get imgs with camera resolution: 2560x720@30Hz

### Actual Result

Failed to get imgs, with error log:
`
[sl_oc::video::VideoCapture] INFO: ZED Open Capture - Camera module - Version: 0.6.0

[sl_oc::video::VideoCapture] INFO: Camera resolution: 2560x720@30Hz

[sl_oc::video::VideoCapture] INFO: Trying to open the device '/dev/video0'

[sl_oc::video::VideoCapture] INFO: Opened camera with SN: 38085162

[sl_oc::video::VideoCapture] ERROR: Error setting the camera resolution

Cannot open camera video capture

See verbosity level for more details.
`

### ZED Camera model

ZED2i

### Environment

```shell
Ubuntu 20.04
AMD
```

### Anything else?

no

Myzhar, 2023-06-30:
Hi @zhukao 
is the camera connected to a USB3 or USB2 port?
Can you copy and paste the outpost of the commands `lsusb -d 2b03:` and `lsusb -d 2b03: -v -t`?

zhukao, 2023-06-30:
> 
Yes, the camera should connect to the USB3 port!
This point should be emphasized in the README file.

This issue is resolved and can be closed.

closed 2023-06-30 by Myzhar