stereolabs/zed-open-capture #37: Availability of raw data

tedyiningding, 2023-07-06:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, bug report, or anything other than a feature request directly related to this project.

### Proposal

Hi.

There seems to be an inconsistency between the following webpage and the description of this repository with regard to the raw data availability.

https://support.stereolabs.com/hc/en-us/articles/1500009136321-Do-the-ZED-cameras-support-capturing-and-saving-raw-data-of-RGB-sensor-e-g-bayer-pattern-dataset-without-nonlinear-processing-
This page says that the raw data is not available, whereas the description of this repository says
> The open-source library provides methods to access raw video frames, calibration data, camera controls, and raw data from the camera sensors (on ZED 2, ZED 2i, and ZED Mini).

Could you please clarify the availability of raw data (e.g. bayer pattern dataset without nonlinear processing)? I am using ZED 2i.

Many thanks,
Ted

Myzhar, 2023-07-06:
Hi @tedyiningding 
with "raw" data we mean the not rectified image information in YUV422 format.
ZED Open Capture only provides this type of stream.
Bayer filtering is performed onboard by the ISP

tedyiningding, 2023-07-06:
Hi @Myzhar 
Thanks for the clarification.

closed 2023-07-06 by tedyiningding