stereolabs/zed-open-capture #38: Sensor_example working but Fails to run the video_example on zed2i

Mandarhk, 2023-07-26:
### Preliminary Checks

- [X] This issue is not a duplicate. Before opening a new issue, please search existing issues.
- [X] This issue is not a question, feature request, or anything other than a bug report directly related to this project.

### Description

I am using Zed2i over Linux 22.04 over VM(VM workstation pro)
After the successful build of all the examples and the library 
I can only able to run sensor_example and produces the sensor data.
but it fails to produce and gives output 
"Cannot open camera video capture" 
"See verbosity level for more details"

### Steps to Reproduce

1.Build the package 
2. try to run video_example
3.
...

### Expected Result

I want to run the video example and later also need to get the data in the ros2 humble . Thank You!

### Actual Result

[image]
[image]
[image]

### ZED Camera model

ZED2i

### Environment

```shell
OS: Linux 22.04 (Over VM)
CPU: x86-64
GPU: No GPU(Cannot access through VM)
ZED SDK: Using open capture
```

Myzhar, 2023-07-26:
Hi @Mandarhk 
can you open the camera with a webcam application like `cheese` or `guvcview`?
Can you copy and paste the output of the command `$ ls /dev/video*`?
Did you enable USB3 in your VM?

Mandarhk, 2023-07-26:
Thank you for your prompt reply!
I tested with guvcview and /dev/video was not found 
later I tried changing USB3 setting in my VM but it was still not working.

But it finally worked in another VM with Linux 20. Is there any package support for the ROS2 wrapper

Myzhar, 2023-07-26:
I'm sorry but we do not provide a ROS 2 Wrapper for the Open Capture driver, only for the full ZED SDK.

Mandarhk, 2023-07-26:
Thank you for your help!

closed 2023-07-26 by Mandarhk